stawel/cheali-charger #318: Turnigy Accucel 6 50W 6A flashing errors.

SumolX, 2022-08-01:
Unable to flash the following files:
unstable/cheali-charger-Turnigy-Accucel-6-50W-6A_2.01-e10.3.12-20170311_atmega32.bin
unstable/cheali-charger-Turnigy-Accucel-6-50W-6A_2.01-e10.3.12-20170311_atmega32.hex
cheali-charger-Turnigy-Accucel-6-50W-6A_2.00-e10.3.12-20161016_atmega32.hex

Successfully flashed:
cheali-charger-Turnigy-Accucel-6-50W-6A_2.00-e10.3.12-20161016_atmega32.bin

I did notice the file size did vary... for example:
32K = cheali-charger-Turnigy-Accucel-6-50W-6A_2.00-e10.3.12-20161016_atmega32.bin (This one was successful)
209K = unstable/cheali-charger-Turnigy-Accucel-6-50W-6A_2.01-e10.3.12-20170311_atmega32.bin

754K = cheali-charger-Turnigy-Accucel-6-50W-6A_2.00-e10.3.12-20161016_atmega32.hex
741K = unstable/cheali-charger-Turnigy-Accucel-6-50W-6A_2.01-e10.3.12-20170311_atmega32.hex

I also was able to backup the eeprom/flash/hfuse/lfuse on my charger.

fhvyhjriur, 2022-08-18:
> I also was able to backup the factory eeprom/flash/hfuse/lfuse on my charger.
What is the size of the backup?

SumolX, 2022-08-18:
eeprom.bin 1KB
flash.bin 32KB
hfuse.bin 1KB
lfuse.bin 1KB