stawel/cheali-charger #324: I am looking for the original firmware Turnigy Accucel 6 50W 6A.

A-S-T-R, 2022-10-14:
Thanks to the developer for the firmware and it really does a lot. I bought a different charger so I decided to go back to the original firmware but found that the backup I made was incorrect. The device is not working.
Now I flashed the firmware previously posted in the developer's archives, these are files from Imax B6. The firmware works, but it seems to me not quite correct, it underestimates the discharge current. I also see that there is not enough calibration limit. I continue to look for the original firmware from Turnigy Accucel 6 50W 6A with the board shown in the photo and I will be grateful if I can get help in this matter.
[image]
[image]

diggit, 2022-10-14:
Hi, this is probably not best place to get original FW support. Try Google group linked in readme. Maybe you will find it there or somebody can rip it.

A-S-T-R, 2022-10-14:
Thanks for the advice. I will delete the question.