stawel/cheali-charger #326: Stable version for original skyrc imax b6 atmega32

atharvbadkas, 2022-11-04:
Hello,
In hex folder, firmwares seems to be build 6-7 years before but I see some changes made recently.
Whats the latest stable version for original imax b6?
If hex not available, Is there any major improvement made in source code?If yes, so I can try to build hex from new version.
I could check the code myself but lack of comments discourages me from doing it.
Im not into coding, just know some basic stuff.

Thanks

TBAMax, 2022-11-10:
As much as I can tell, there are no major changes to the atmega32. Looking at the history of commits. Only mostly additions for new processor support, and such. It is quite ready already those years ago, and highly functional to the smallest detail, however the code is not easy to read, it takes a  quite genius to dissassemble its guts or to add something to it further...

atharvbadkas, 2022-11-13:
Okay...thanks

closed 2022-11-13 by atharvbadkas