stawel/cheali-charger #333: Imax B6 80W clone compatability?

hojnikb, 2023-03-11:
Do these recent clones from aliexpress for example still support cheali? Or are they using different, unsupported MCUs?

joshgarde, 2023-03-11:
Depends on the clone; can't tell by just looking at the listings. In general though, the ones that advertise 32-bit CPUs are safer bets than just going for the cheapest possible listing because the chances its a reprogrammable ARM are higher

silvio2402, 2023-04-18:
Hi, I hope you are doing well.
I bought such a B6 80W clone from [here](https://www.aliexpress.com/item/1005005028773427.html) on AliExpress. It has a [CMS32L051](https://www.mcu.com.cn/uploads/img1/tupian/手册下载/CMS32L051_datasheet_V1.5.pdf).
Would this be compatible? And would 80W theoretically work with the cheali firmware?
I know 80W is a BS number because the included power supply only outputs $12V \times 5A = 60W$.
Thanks

diggit, 2023-04-19:
Hi @silvio2402, this MCU is currently not supported. It's an ARM with enough storage, so it's definitely doable. The only issue I see is reference manual in chinese. Then someone has to add support for it (C++ coding).

silvio2402, 2023-04-19:
Hi @diggit, I have some electronics and coding knowledge and I'd be willing to invest some time to collaborate to make this work. Is [this](https://www.mcu.com.cn/uploads/img1/tupian/2021/CMS32L051%E7%94%A8%E6%88%B7%E6%89%8B%E5%86%8C_V1.2.1.pdf) the reference manual you're referring to?
Also, are there any resources available from the cheali-charger project which files need to be created/edited? If you could mention some commits/pull requests which could help me understand what it takes to support another MCU it would be very helpful. I'll have to get familiar with the codebase so is there any documentation?
If I need help, is the Google Group an appropriate place to ask about the development?
Thanks

diggit, 2023-04-19:
Yes, that is reference manual I've found. Maybe it's not exactly necessary to study it, as there is also some code in that keil pack on their website (it's just zip archive).

Regarding cheali development
I did not contribute any code here :sweat_smile:.
You'll need new directory here https://github.com/stawel/cheali-charger/tree/master/src/hardware that includes some drivers and hw configurations. Look into those already there. Then some changes to cmake configs https://github.com/stawel/cheali-charger/blob/master/CMakeLists.txt etc.

About development tools...
I use VSCode/VSCodium with cortex-debug and cmake plugins (some other plugins for higher comfort).
As compiler: GCC (arm-none-eabi) works well.
STlink/J-Link as SWD probes, but there are other options...

I do this stuff as daily job, so I can also have a look, if you need help. Why everyone has to copy most ugly charger in the world? I'll probably also get one anyway, for science...

silvio2402, 2023-04-19:
Thanks for your reply!

I've downloaded and unpacked the keil pack from the Cmsemicon site and copied the `CMSIS`, `Device`, and `Driver` folders to a new folder at `src\hardware\cmsemicon-CMS32L051\cpu\CMSIS` and I've copied and adjusted the `cmsis.cmake` from the `nuvoton-M051` cpu present in this repo.

However, some CMSIS drivers/files are missing. From the files included in the `nuvoton-M051` `cmsis.cmake` file, the following files are missing in the `cmsemicon-CMS32L051` CMSIS pack: `timer.c`, `uart.c`, `pwm.c`, `FMC.c`, `sys.c`, and `_syscalls.c`. I suppose I have to write them myself?

Also, the `CMSIS\Include\core_cm0plus.h` file from the pack tries to include `cmsis_version.h` and `cmsis_compiler.h`, but they're missing from the pack. Can I copy them from the here: [ARM-software/CMSIS_5/CMSIS/Core/Include](https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/Core/Include)?

Your job sound really cool btw!
I really appreciate your time.

RadioControl, 2023-05-04:
Hi,

i purchased also a B6 clone with a CMS32L051 chip. Hopefully there will be a solution for these ones too!...
[image]
[image]

RadioControl, 2023-05-08:
Hi Silvio,

were you able to move forward with this CMS32L051 project?...

RadioControl, 2023-05-09:
I think a lot of CMS32L051 owner would appreciate that they could use Cheali on their charger. If You were here in the near i would offer mine for the the project. The charger cost about 20 euros, but if someone claim a refund like i did due to misleading description (see pic), it cost just the half...
[image]

diggit, 2023-05-14:
@RadioControl where did you get yours?

RadioControl, 2023-05-14:
@diggit Is it allowed to provide the link?...

RadioControl, 2023-06-01:
Received my new [clone](https://www.aliexpress.com/item/32800729065.html?spm=a2g0o.order_list.order_list_main.65.21381802h9x8Aw) with a nice Atmega32A inside...
[image]
[image]

silvio2402, 2023-06-01:
Cool! Do you mind sharing the purchase link?
Btw I haven’t had the time to develop it further for the other chip.

RadioControl, 2023-06-01:
Just click on "clone" ;) ...Diggit found out that the CMS ones have not the best hw #340. I charged some 3 cell lipos with mine CMS clone in balance mode. I think it's not that bad. Other modes i didn't try.