stawel/cheali-charger #334: help me

zuhair01, 2023-04-11:
i have b6ac pro board v: b65A-A-V.17 charger i update the software but cannot access the keypad and thq
[image]
[image]

stawel, 2023-04-15:
> randomly beep

this is something I haven't seen before,
could you make a video when powering on the charger?

> avrdude.exe: safemode: Fuses OK (E:FF, H:C9, L:2F)

you probably should "reset atmega32 fuses", see https://github.com/stawel/cheali-charger#troubleshooting

> cannot access the keypad.

if only keypad is not working (and probably the buzzer)
check to what the keypad pins are connected to - to which atmega32 pins

zuhair01, 2023-04-16:
keypad pins are connected to
p.no 2- start
pno 3 - inc
p.no 4 - dec
p.no 5 - batt type

https://user-images.githubusercontent.com/24286542/232318739-4ba28cee-5434-49f7-8274-23aa59ba3eb3.mp4

stawel, 2023-04-17:
Thank you for the video,

ok the pins are as they should:
https://github.com/stawel/cheali-charger/blob/master/src/hardware/atmega32/generic/50W/imaxB6-pins.h#L78-L81

I'm guessing (??)  that Your charger is missing pull-up resistors on the keypad - they are not needed but cheali-charger
assumes they are present, to confirm that please do a test:
1. measure voltages on each pin on keypad side for:
"start"
"inc"
"dec"
"batt type"
2. measure the voltage as above but this time when the corresponding button is pressed

zuhair01, 2023-04-17:
voltages on each pin on keypad side for:
"start" - 4.88v
"inc" - 4.88v
"dec" - 0.12v
"batt type" - 4.88v

button is pressed
"start" - 0v
"inc" - 0v
"dec" - 0v
"batt type" - 0v

zuhair01, 2023-04-17:
dec key left side 4.8V other side 0V
[image]

stawel, 2023-04-19:
Something is definitely broken, possibilities:
p1. a damaged PCB via: https://en.wikipedia.org/wiki/Via_(electronics)
p2. PCB traces shorted
p3. your PCB has additional components (connected to "dec" pin)   which are damaged 
p4. atmega32 is damaged

if you measured the 0.12V exactly at the 10k resistor then it is definitely not p1.

for p4. you may try this:
[cheali-charger-imaxB6-original_2.01-e10.3.12-20230419_atmega32-ENABLE_KEYBOARD_PULLUP_RESISTORS.zip](https://github.com/stawel/cheali-charger/files/11277197/cheali-charger-imaxB6-original_2.01-e10.3.12-20230419_atmega32-ENABLE_KEYBOARD_PULLUP_RESISTORS.zip)
(version with  pull-up resistors enabled on keypad pins)
If your charger works with this version that means your atmega32 is definitely damaged, you should replace it. 
(if it is damaged on this pin there is a good chance it is  also damaged somewhere else)

zuhair01, 2023-04-20:
not a change,. 
your firmware start,inc,
and batt type keys make beep  . I found firmware Google search is firmware are the 3 keys work still dec key not working
[ImaxMod.zip](https://github.com/stawel/cheali-charger/files/11287530/ImaxMod.zip)

zuhair01, 2023-05-01:
old firmware backup copy i attach 
Thanks for your help
[b6ac pro.zip](https://github.com/stawel/cheali-charger/files/11366143/b6ac.pro.zip)

diggit, 2023-05-01:
Yep, readout protection was active an your flash (an eeprom?) backup is useless.

diggit, 2023-05-01:
It really sounds like a HW issue. Do you remember if all keys worked in original FW?

zuhair01, 2023-05-01:
yes all key I found firmware Google search is firmware are the 3 keys work still

zuhair01, 2023-05-01:
cheali firmware no key work