stawel/cheali-charger #335: B6AC original - target does not answer

LeonHartStrife, 2023-04-28:
I updated an original B6AC with an USBASP and AVRDUDE and got some squares on the first line of the screen. 
I pluged the charger and since then I have a black screen and the message "target does not answer". 
I still can update the charger if I put "-F" in AVRDUDE but it won't turn on if I plug in. 
Is the charger dead or bricked?

diggit, 2023-04-28:
Hi,
did you follow flashing guide https://github.com/stawel/cheali-charger/blob/master/docs/flashing.md ?
During backup, has avrdude detected correct MCU?
What are your fuse values?

Squares on display are default state after power up of display. It means that display is not being initialized and thus FW is probably not running.

diggit, 2023-04-28:
Also please attach avrdude logs as txt file or as code block.

LeonHartStrife, 2023-04-29:
Hi diggit, thanks for your help. 
I did follow the flashing guide but I forgot to save the hex files.
I backup the bin files succefully and got four bin files :
- eeprom.bin (1ko)
- flash.bin (32ko)
- hfuse.bin (1ko)
- lfuse.bin (1ko)

Then I flashed the charger with cheali-charger-imaxB6-original_2.00-e10.3.12-20160613_atmega32.hex and got those squares. 
Thinking the charger was well flashed, I plugged it AC and the screen turned on a sec (with square) and turned off. 

My first error I presure was not to save the original hex files and my second one was to flash it with the wrong firmware. 

Now, when I try to flash it, here is what I got from AVRDUDE : 

```
C:\Users\LeonH\Desktop\AVRDUDE>avrdude.exe -patmega32 -cusbasp -Uflash:w:cheali-charger-Dual-Power-B6AC-80W-RC_2.00-e10.3.12-20160613_atmega32.hex:a

avrdude error: cannot set sck period; please check for usbasp firmware update
avrdude error: program enable: target does not answer (0x01)
avrdude error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower ISP clock frequency, e.g. -B 125kHz
        - use -F to override this check

avrdude done.  Thank you.

C:\Users\LeonH\Desktop\AVRDUDE>avrdude.exe -patmega32 -cusbasp -Uflash:w:cheali-charger-Dual-Power-B6AC-80W-RC_2.00-e10.3.12-20160613_atmega32.hex:a -F

avrdude error: cannot set sck period; please check for usbasp firmware update
avrdude error: program enable: target does not answer (0x01)
avrdude error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower ISP clock frequency, e.g. -B 125kHz
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x000000 (retrying)
avrdude: device signature = 0x000000 (retrying)
avrdude: device signature = 0x000000
avrdude error: Yikes!  Invalid device signature.
avrdude warning: expected signature for ATmega32 is 1E 95 02
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.

avrdude done.  Thank you.

C:\Users\LeonH\Desktop\AVRDUDE>avrdude.exe -patmega32 -cusbasp -Uflash:w:cheali-charger-Dual-Power-B6AC-80W-RC_2.00-e10.3.12-20160613_atmega32.hex:a -F -B 125kHz

avrdude: set SCK frequency to 93750 Hz
avrdude error: cannot set sck period; please check for usbasp firmware update
avrdude error: program enable: target does not answer (0x01)
avrdude error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower ISP clock frequency, e.g. -B 125kHz
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x000000 (retrying)
avrdude: device signature = 0x000000 (retrying)
avrdude: device signature = 0x000000
avrdude error: Yikes!  Invalid device signature.
avrdude warning: expected signature for ATmega32 is 1E 95 02
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.

avrdude done.  Thank you.
```

LeonHartStrife, 2023-04-29:
Here's what I got when I use the command "-F" and "-B 128kHz"
If I don't put these commands, flashing won't start with the message : "target does not answer"

```
C:\Users\LeonH\Desktop\AVRDUDE>avrdude.exe -patmega32 -cusbasp -Uflash:w:cheali-charger-Dual-Power-B6AC-80W-RC_2.00-e10.3.12-20160613_atmega32.hex:a -F -B 128kHz

avrdude: set SCK frequency to 93750 Hz
avrdude error: cannot set sck period; please check for usbasp firmware update
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9502 (probably m32)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: set SCK frequency to 93750 Hz
avrdude error: cannot set sck period; please check for usbasp firmware update
avrdude: reading input file cheali-charger-Dual-Power-B6AC-80W-RC_2.00-e10.3.12-20160613_atmega32.hex for flash
         with 32768 bytes in 1 section within [0, 0x7fff]
         using 256 pages and 0 pad bytes
avrdude: writing 32768 bytes flash ...

Writing | ################################################## | 100% 20.84 s

avrdude: 32768 bytes of flash written
avrdude: verifying flash memory against cheali-charger-Dual-Power-B6AC-80W-RC_2.00-e10.3.12-20160613_atmega32.hex

Reading | ################################################## | 100% 13.80 s

avrdude: 32768 bytes of flash verified

avrdude done.  Thank you.

C:\Users\LeonH\Desktop\AVRDUDE>
```

stawel, 2023-04-29:
> I did follow the flashing guide but I forgot to save the hex files.
> I backup the bin files succefully 

bin and hex files contain the same information

> Where can I find my fuse values ?

Your current fuse values should be in (you can view them with a hex editor)
* hfuse.bin (1ko)
* lfuse.bin (1ko)

but I would try this: https://github.com/stawel/cheali-charger#troubleshooting

LeonHartStrife, 2023-04-29:
Thanks for your help guys. =)

Here are my fuse values : 

- hfuse.bin = C9
- lfuse.bin = 3F

I tried to reflash with the saved flash.bin file, flashing seems good in avrdude but the charger does not turn on (screen is off).

stawel, 2023-04-29:
> I tried to reflash with the saved flash.bin file ... but the charger does not turn on 

there is a good chance that your flash was "locked", you can check this by opening the flash.bin 
in a hex editor: if you see a pattern: consecutively increasing numbers (as far as I remember) 
then the flash was locked.

>Here are my fuse values :
> * hfuse.bin = C9
> * lfuse.bin = 3F

https://eleccelerator.com/fusecalc/fusecalc.php?chip=atmega32a&LOW=3F&HIGH=C9&LOCKBIT=FF

definitely I would look into: https://github.com/stawel/cheali-charger#troubleshooting
(I've just edited it, I hope it is more readable now)

LeonHartStrife, 2023-04-30:
> there is a good chance that your flash was "locked", you can check this by opening the flash.bin
> in a hex editor: if you see a pattern: consecutively increasing numbers (as far as I remember)
> then the flash was locked.

I can indeed see a pattern with increasing numbers and characters.

> definitely I would look into: https://github.com/stawel/cheali-charger#troubleshooting

I'm totally a noob so here's what I've done : 

> 1. After flashing I see "options" and some strange characters in the second line (for example: squares), what should I do?
> reset the charger to default settings (go to: "options"->"reset default" and press the "start" button)

I can't reset to default as the screen is off.

> 2. I get "calib. error": see [this.](https://github.com/stawel/cheali-charger/blob/master/docs/calibration_error_codes.md)

Same as first point = screen is off, no message.

> 3. Occasionally when the power is turned on, the error "eeprom error: 2 yes" pops up (or any other number): see https://github.com/stawel/cheali-charger/issues/291

Same.

> 4. After flashing charger doesn't work (display shows squares) : download the *.hex again, use the "RAW" button in github

To be sure what I have to do : I used the RAW button, then right click and "save as". I get a .txt file that I rename to .bin or .hex file. I have downloaded the four files this way. Then : 

> check the sha1 sum of the file, compare it with *.sha1: 

I compared the sha1 sum of the .hex file with the .hex.sha1's one and get the same number for both files

- For cheali-charger-dual-power-b6ac-80w-rc_2.00-e10.3.12-20160613_atmega32.hex = 4360aa3b52db1c71aacf9880310bc23e52c62bef
- For cheali-charger-Dual-Power-B6AC-80W-RC_2.00-e10.3.12-20160613_atmega32.hex.sha1 = 
4360aa3b52db1c71aacf9880310bc23e52c62bef

> 5. [Atmega32 CPU] Sha1 sum is correct and the charger still doesn't work (display shows squares):
> reset atmega32 fuses using avrdude:
> windows: avrdude.exe -patmega32 -cusbasp -Uhfuse:w:0xc5:m -Ulfuse:w:0x3f:m

I did reset the fuses with these values and also tried with the original values. 

I started the flashing process : 

```
>>>: avrdude -c usbasp -p m32 -B 0.5 -U flash:w:"C:\Users\LeonH\Desktop\AVRDUDE\cheali-charger-Dual-Power-B6AC-80W-RC_2.00-e10.3.12-20160613_atmega32.hex":a 

avrdude.exe: set SCK frequency to 1500000 Hz
avrdude.exe: warning: cannot set sck period. please check for usbasp firmware update.
avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e9502 (probably m32)
avrdude.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
avrdude.exe: erasing chip
avrdude.exe: set SCK frequency to 1500000 Hz
avrdude.exe: warning: cannot set sck period. please check for usbasp firmware update.
avrdude.exe: reading input file "C:\Users\LeonH\Desktop\AVRDUDE\cheali-charger-Dual-Power-B6AC-80W-RC_2.00-e10.3.12-20160613_atmega32.hex"
avrdude.exe: input file C:\Users\LeonH\Desktop\AVRDUDE\cheali-charger-Dual-Power-B6AC-80W-RC_2.00-e10.3.12-20160613_atmega32.hex auto detected as Intel Hex
avrdude.exe: writing flash (31308 bytes):

Writing | ################################################## | 100% 20.00s

avrdude.exe: 31308 bytes of flash written
avrdude.exe: verifying flash memory against C:\Users\LeonH\Desktop\AVRDUDE\cheali-charger-Dual-Power-B6AC-80W-RC_2.00-e10.3.12-20160613_atmega32.hex:
avrdude.exe: input file C:\Users\LeonH\Desktop\AVRDUDE\cheali-charger-Dual-Power-B6AC-80W-RC_2.00-e10.3.12-20160613_atmega32.hex auto detected as Intel Hex

Reading | ################################################## | 100% 14.98s

avrdude.exe: 31308 bytes of flash verified

avrdude.exe done.  Thank you.
```

And after that, the screen remains off.
I'm going to post a video to show what is happening during the flashing.

LeonHartStrife, 2023-04-30:
Here it is, hope this helps. 
https://youtu.be/o2qxhZ_fk9Y

stawel, 2023-04-30:
yes points 4 and 5 (from [troubleshooting](https://github.com/stawel/cheali-charger#troubleshooting)) were meant for you.

Unfortunately I'm a non believer when it comes to point 5:

> I did reset the fuses with these values and also tried with the original values.

please run a command in command line that looks similar to this:
```
# update fuses to: 0xc5, 0x3f
avrdude -c usbasp -p m32 -B 0.5 -Uhfuse:w:0xc5:m -Ulfuse:w:0x3f:m

# save the current state of fuses, lock, calibration
avrdude -c usbasp -p m32 -B 0.5 -Ulfuse:r:lfuse.hex:i -Uhfuse:r:hfuse.hex:i  -Ulock:r:lock.hex:i -Ucalibration:r:calibration.hex:i
```
and post the output logs and *.hex files here.

LeonHartStrife, 2023-04-30:
Here's what I got :

```
C:\Users\LeonH\Desktop\AVRDUDE>avrdude.exe -c usbasp -p m32 -B 0.5 -Uhfuse:w:0xc5:m -Ulfuse:w:0x3f:m

avrdude: set SCK frequency to 1500000 Hz
avrdude error: cannot set sck period; please check for usbasp firmware update
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9502 (probably m32)
avrdude: reading input file 0xc5 for hfuse
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte hfuse ...
avrdude: 1 byte of hfuse written
avrdude: verifying hfuse memory against 0xc5
avrdude: 1 byte of hfuse verified
avrdude: reading input file 0x3f for lfuse
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte lfuse ...
avrdude: 1 byte of lfuse written
avrdude: verifying lfuse memory against 0x3f
avrdude: 1 byte of lfuse verified

avrdude done.  Thank you.

C:\Users\LeonH\Desktop\AVRDUDE>avrdude.exe -c usbasp -p m32 -B 0.5 -Ulfuse:r:lfuse.hex:i -Uhfuse:r:hfuse.hex:i  -Ulock:r:lock.hex:i -Ucalibration:r:calibration.hex:i

avrdude: set SCK frequency to 1500000 Hz
avrdude error: cannot set sck period; please check for usbasp firmware update
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9502 (probably m32)
avrdude: reading lfuse memory ...
avrdude: writing output file lfuse.hex
avrdude: reading hfuse memory ...
avrdude: writing output file hfuse.hex
avrdude: reading lock memory ...
avrdude: writing output file lock.hex
avrdude: reading calibration memory ...
avrdude: writing output file calibration.hex

avrdude done.  Thank you.
```

Here are the files : 
[logs files.zip](https://github.com/stawel/cheali-charger/files/11361543/logs.files.zip)

Let me know if you prefer another type of file instead of zip.
Thanks again for your much apreciated help.

TBAMax, 2023-04-30:
Hi. Have not much time to dig into the conversation. But just for info I have had good result with the lfuse:0xAF and hfuse:0xC9.

LeonHartStrife, 2023-05-02:
@TBAMax Hi, thanks for your answer.

Just tried to flash with your fuse values with no luck unfortunately.

stawel, 2023-05-04:
@LeonHartStrife 

ok, I'm beginning to suspect that the schematics of your board has changed
we will need to do some investigations.

first question: Why your display goes "black" after programming?
normally the programmer should give enough current to power the display.

please do:
1. make sure you correctly connected the Vcc and GND pins to the programmer
2. measure the Vcc voltage during programming and after 

I assume we do something in software that drains the Vcc current:

3. please test all *.hex files in: [charger-tests.zip](https://github.com/stawel/cheali-charger/files/11397396/charger-tests.zip) and post which of the files turned the display off.

later  you probably will have to change the LCD connector so we can see what is below,
one possibility is to use 2.54mm pins, something like this:
[image]

diggit, 2024-02-28:
@LeonHartStrife did you have any luck? Are you planning/willing to investigate it more?