stawel/cheali-charger #336: HK ECO-6-8 200W

RadioControl, 2023-05-04:
Hi folks,

i'm planning to flash this charger with the cheali-charger-Turnigy-A-6-10-200W_2.01-e10.3.12-20170311_atmega32.hex.hex
Unfortunately the flash and eeprom read outs are fake ones.
In the attached pic you can see the read bits. Do i have to change these before flash the hex file? Especially the 0x38 lock bit?
Thanks in advance!...
[image]

stawel, 2023-05-04:
according to:
https://ww1.microchip.com/downloads/en/DeviceDoc/Atmega32A-DataSheet-Complete-DS40002072A.pdf
page: 261
```
27.7.3 Chip Erase
The Chip Erase will erase the Flash and EEPROM(1) memories plus Lock bits. The Lock bits are not reset until the
program memory has been completely erased. The Fuse bits are not changed. A Chip Erase must be performed
before the Flash and/or the EEPROM are reprogrammed.

Note: 1. The EEPRPOM memory is preserved during chip erase if the EESAVE Fuse is programmed.
```

The lock bits will be erased automatically when you flash the device - the "Chip Erase" command will be performed  by `avrdude` before flashing.

RadioControl, 2023-05-05:
Thank You for this! "the "Chip Erase" command will be performed by avrdude before flashing." In Avrdudess as you can see in the pic there is a "Erase flash and EEPROM" option. Do i have to check this or is it unnecessary?
Where is the calibration saved, in eeprom or in the flash?

RadioControl, 2023-05-05:
In meanwhile the flash went successful. Noe it's ready for calibration...
[image]
[image]

RadioControl, 2023-05-07:
...At the 50mA calibration it blows the 250mA fuses all the time. Why?...

diggit, 2023-05-07:
Default calibration point may be way off. Try starting with higher range of your Ammeter, then dial the current down to 50 mA a then switch to some lower range of your Ammeter.

RadioControl, 2023-05-08:
Thanks for your reply, but i think the problem is that when i hit the start button, the current starts at higher (maybe around 850mA) for a second or so and after that goes down to the lower amps. What is the reason for that?...

RadioControl, 2023-05-14:
I'm struggling with the Icharge calibration. What i find out is when i start the 50mA (whatever value is set) calibration starts for a moment at ca. 850mA (enough to blow the multimeter 200mA fuse). When i start with the 20A port on the DMM and dial the current below 200mA to plug the multimeter wire to the mA port, the calibration current flow stops. So i tricked this with shorting like in the pic...My questions are: What is the best way now to determine the bottom current limit of my charger? What is to set with the "Iexpect" value? If i found the bottom current limit what do i have to set in other options of the menu?
[image]

diggit, 2023-05-14:
I can confirm such behavior on my Turnigy 400Wx2.
Waveforms below were observed when current calibration range is highlighted (Value nor whether charging or discharging does not matter.) and **start** button is pressed (Current should not be flowing yet.).
Measured with 11.1 V battery on output.
Same test with 15.6 V battery produced same waveforms. So it does not seem to be voltage dependent.

There are 2 issues
- 50 ms discharge pulse with approximately 950 mA flowing
- very short discharge peak around 4.7 A

Any thoughts @stawel ?
Observed discharge pulse may be even intentional, bu I can't think of any reason.

[image]
[image]
[image]

stawel, 2023-05-19:
hm... interesting! ... but as an nonbeliever :-) I have to take out my own oscilloscope

stawel, 2023-05-19:
R138, R150    measured
in: https://drive.google.com/file/d/0B1RXXTatsA1cb1R5NHM3MEtsakE/edit?resourcekey=0-YuNi1Kod3xXI4xazR8Gb_Q
 during calibration start, scale is wrong but it confirms your findings!

[image]

ok I'm opening a new issue for that

stawel, 2023-05-19:
@RadioControl 

> My questions are: 
> What is the best way now to determine the bottom current limit of my charger?

set "value:" to the lowest value (probably 1), the lowest possible current should be visible on your DMM
 
> What is to set with the "Iexpect" value? If i found the bottom current limit what do i have to set in other options of the menu? 

> How to change default calibration point?... 

1. set "I expect:" to your desired current calibration point, (example 850mA)
2. go to "value: " change it until you see on DMM the "I expected:" current (example 850mA)
3. IMPORTANT! press: "OK" in the "value:" line - this will save your calibration point including "Iexpect:"

additional note:
picking "bottom current limit" as the first calibration point may not be the best choice, we do a two point
linear interpolation (or extrapolation as some people call it) which means your current measurements will
be most accurate near these points and 
I doubt you will ever charge a battery with a current near ~0 mA, also the charging current near its bottom limit starts to behave less linear

stawel, 2023-05-21:
@RadioControl  
could you try a version from: https://github.com/stawel/cheali-charger/tree/master/hex/unstable
it should stop burring your fuses during calibration.

RadioControl, 2023-05-22:
@stawel 
i will try it asap, spare fuses are on the way to me. Thank You!...

RadioControl, 2023-07-10:
> @RadioControl could you try a version from: https://github.com/stawel/cheali-charger/tree/master/hex/unstable it should stop burring your fuses during calibration.

@stawel 
i tried the "cheali-charger-Turnigy-A-6-10-200W_2.02-e10.3.12-20230523_atmega32" fw but the issue is still exist. Start calibration with a short current of ca. 700mA waht is blowing the fuse...The closest stable minimum charge current to calibrate is about 140mAh. All below that has a too long time to start the charge. I guess it's a hardware depending thing but i will try the "cheali-charger-GTPowerA6-10_2.02-e10.3.12-20230523_atmega32" hex file if there are any differences...

RadioControl, 2023-07-10:
Another issue i found out is that on the fan connection is about 12V without the the fan connected but with the fan connected it is about 2,5V. What cause this?...

RadioControl, 2023-10-14:
> Another issue i found out is that on the fan connection is about 12V without the the fan connected but with the fan connected it is about 2,5V. What cause this?...

@stawel @diggit Can You please help me with this?...

diggit, 2023-11-01:
Hi,
Check the fan, if it spins on some PSU. It's 5 V fan, so even USB could be enough.
Other thing could be damaged fan switching transistor.
When you connect fan, is the charger running without issues? That would mean issue with mentioned transistor.