stawel/cheali-charger #337: Current (1A) is flowing when entering calibration menus

stawel, 2023-05-19:
reported:
https://github.com/stawel/cheali-charger/issues/336#issuecomment-1537285964
https://github.com/stawel/cheali-charger/issues/336#issuecomment-1546863346

@diggit:
Waveforms below were observed when current calibration range is highlighted (Value nor whether charging or discharging does not matter.) and start button is pressed (Current should not be flowing yet.).
Measured with 11.1 V battery on output.
Same test with 15.6 V battery produced same waveforms. So it does not seem to be voltage dependent.

There are 2 issues

- 50 ms discharge pulse with approximately 950 mA flowing
- very short discharge peak around 4.7 A

[image]

stawel, 2023-05-19:
> 50 ms discharge pulse with approximately 950 mA flowing

from my understanding this issue is caused by:
https://github.com/stawel/cheali-charger/blob/master/src/core/calibration/CurrentCalibrationMenus.cpp#L114

`Program::dischargeOutputCapacitor()` should only discharge the output capacitor, not 
the battery.

> very short discharge peak around 4.7 A

this needs more investigation

diggit, 2023-05-19:
I believe that discharging circuit is never fully disconnected due to body diode in Q22 ([schematic](https://drive.google.com/file/d/0B1RXXTatsA1cb1R5NHM3MEtsakE/edit?resourcekey=0-YuNi1Kod3xXI4xazR8Gb_Q))

So it's not possible to discharge output capacitor without current draw from battery.

diggit, 2023-05-19:
Attempting to discharge C22/C23 will always draw current from battery, which is connected right to it. Am I missing something?

I can't see a reason in attempting to do so.

stawel, 2023-05-21:
@diggit 

> Attempting to discharge C22/C23 will always draw current from battery, which is connected right to it. Am I missing something?

I missed it 

> I can't see a reason in attempting to do so.

Yes, It doesn't make sense to me  either now, and I don't remember why I've added it.
(probably  I should write better commit messages in the future)

Ok, I've removed the `dischargeOutputCapacitor()` code, 
Thanks for Your measurements, without them I would probably ignore this.

diggit, 2023-07-06:
@stawel 
Maybe there is one reason to attempt discharging at the beginning.
In scenario, when you are starting charging/discharging/..., output voltage is checked. Charger could measure just some residual charge of output capacitor instead of battery voltage and allow charging/discharging to start which would end up in immediate UV/OV condition.
I guess this could be the original reason for short discharging prior any other battery action. 1 A is probably excessive, I'd go with minimum discharge current (battery or charger, whichever is higher) and disable it in calibration.