stawel/cheali-charger #339: Inaccurate voltage for cells 1 and 2 when charging

intelligentpotato, 2023-05-21:
Turnigy Accucell 6 50W running v2.00_20160613. Calibrated each cell to +/- 5 mV @ 3.96 V. When I start charging, the voltage on cells 1 and 2 jumps to 4.20-4.21 V (@ 2.1 A current) according to the charger. In reality it is 4.01 V according to the voltmeter. I've read about the hardware limitation that prevents accurate voltage measurement for cells 1 and 2, but 0.2 V error?! Is my unit faulty (it is indeed old) or am I doing something wrong? Previously I'd been blaming my old packs, but started digging deeper after buying new ones, and it turns out to be the problem with the charger itself.
Additionally, please share the original firmware if you have one. I've lost mine after reinstalling the OS.
Thanks!

RadioControl, 2023-05-22:
Hi, i have a read out from my Accucell6 50W before the Atmega32L were blocked somehow (restore in progress with hvpp). I never tried to reflash the stock fw and i don't know if it is a valid read out. Maybe someone experienced can investigate it...
[accucel6-50w-oem.txt](https://github.com/stawel/cheali-charger/files/11526540/accucel6-50w-oem.txt)

intelligentpotato, 2023-05-22:
@RadioControl according to [this](https://groups.google.com/g/cheali-charger/c/0f7FD9yRkvY) your dump is useless :(

intelligentpotato, 2023-05-22:
@RadioControl I found a working backup. Not from this exact model, but it works.
https://groups.google.com/g/cheali-charger/c/pz4mmbuzVj0

Unfortunately, it is also showing 4.16 V for cell 1 while in reality it is 3.96 V. I guess my unit is faulty. Any comments are welcome. I will be closing this issue in a day or two.

RadioControl, 2023-05-22:
Ok, thanks for the info!

stawel, 2023-05-22:
@intelligentpotato 
> Turnigy Accucell 6 50W running v2.00_20160613. 

What is the exact name of the hex file you used?

> upd: in the expert calibration menu I only have Vplus and Vminus

if you don't have "Vb0pin", "Vb1pin", Vb2pin" you have flashed a version with "more advanced voltage measurement on first two cells", see:  [differences-between-versions](https://github.com/stawel/cheali-charger/blob/master/docs/flashing.md#differences-between-versions)
that's a wrong one (or maybe the old versions didn't have one? I don't remember)

I would try this:
1. flash you charger with: [cheali-charger-Turnigy-Accucel-6-50W-5A_2.02-e10.3.12-20230522_atmega32.hex](https://github.com/stawel/cheali-charger/blob/master/hex/unstable/cheali-charger-Turnigy-Accucel-6-50W-5A_2.02-e10.3.12-20230522_atmega32.hex)
2. make sure you have "Vb0pin", "Vb1pin", Vb2pin" in "options"->"calibrate"->"expert DANGER!"
3. reset charger: "options"->"reset default"
4. calibrate charger normally (no need for "expert DANGER!" yet)
5. check voltages during charge
6. if voltages are still off: do ["expert DANGER!"](https://github.com/stawel/cheali-charger/blob/master/docs/calibration_expert.md)
7. check voltages during charge
8. if voltages are still off: hm... maybe your charger is damaged, repair charger: more info on that when 1.-7. fails

intelligentpotato, 2023-05-23:
@stawel oh man, works like a charm! Internal resistance and cell voltage measurement are perfectly on point even without the expert calibration. So it turns out I'd been struggling for months due to my own indiscretion. 
I didn't know there were both 5A and 6A modifications of the Accucell 6. Mine states 10.10 production date on the sticker on the PCB.

closed 2023-05-23 by intelligentpotato

RadioControl, 2023-05-25:
@stawel 
@diggit

can You confirm that my backup is locked?

diggit, 2023-05-25:
Yes , your AVR was locked when you did this backup. You can see patternn in there (ignoring addressing columns).
Compare it visually for exame with this hex file https://github.com/stawel/cheali-charger/blob/master/hex/unstable/cheali-charger-ADCKeyboardAnalyzer-150W_2.02-e10.3.12-20230523_atmega32.hex