stawel/cheali-charger #340: Imax B6 with CMS32L051 status

diggit, 2023-05-26:
This issue is tracking Imax B6 80W with CMS32L051 mcu, reverse engineering and possibly future support.
Origin in #333

# Progress
- MCU pinout reverse engineering: 0 %

# Basic info about charger:
- input voltage: 11-18 Vdc
- max charging power: 80 W
- max discharging power: ?
- charging current: 0.1-6.0 A
- discharging current: 0.1-2.0 A
- max balancing current: ?
- NiCd/NiMH cells: 1-16
- Lixx cells : 1-6
- Lead battery: 2V - 20V

# Findings
- simple 2 layer PCB
- design is mostly the same as other Imax B6 chargers have
- [CMS32L051](https://www.mcu.com.cn/en/Products/291/pids/5.html) MCU (Cortex M0+, 64 MHz, 64 kB FLASH, 8 kB RAM, 1.5 kiB data FLASH?) 
  - [Datasheet](https://www.mcu.com.cn/uploads/img1/tupian/%E6%89%8B%E5%86%8C%E4%B8%8B%E8%BD%BD/CMS32L051_%E6%95%B0%E6%8D%AE%E6%89%8B%E5%86%8C_V1.9.3.pdf)
  - [Ref. manual](https://www.mcu.com.cn/uploads/img1/tupian/2021/CMS32L051%E7%94%A8%E6%88%B7%E6%89%8B%E5%86%8C_V1.2.1.pdf)
- powerstage:
  - switching frequency somewhere between 53.4 kHz
  - horrible ringing on booster stage (slow diode?)
- **no opamps** ?!
- no external analog mux for ADC

diggit, 2023-05-27:
Hi @RadioControl,
Charger arrived. I did quick check of HW and I am quite disappointed.
Cell voltage measurement sucks, because there isn't single op amp for differential cell voltage measurement. Whole cell voltage measurement circuitry consists of dividers and direct connection to MCU with 12 bit ADC. Due to this fact, highest cell measurement has to be able to measure at least 26 V. Resolution is then 6 mV at best (`26 / 4096 = 0.006`).

I am more inclined to focus on #338 now, because for just a few more bucks, you can get better and smaller charger.

RadioControl, 2023-05-27:
Hi @diggit ,

thank You for Your effort and toughts on that!...another [B6 clone](https://www.aliexpress.com/item/32800729065.html?spm=a2g0o.order_list.order_list_main.35.21ef1802PQ4u2v) for a few €s more is on the way to me, but before i placed the order i asked what chip is in there, they answered Atmega32. We'll see...

RadioControl, 2023-06-08:
Hi @diggit ,

did You found any calibration menu on it?... For me no button combination to get in a calibration menu, altough i found that the stock calibration is not that bad, at least with a 3s lipo in balance charging mode...

DSchndr, 2023-12-20:
Hi, the hardware was already reverse engineered [here](https://www-user.tu-chemnitz.de/~heha/basteln/Konsumg%C3%BCter/Ladeger%C3%A4te/)
here is [its schematic](https://www-user.tu-chemnitz.de/~heha/basteln/Konsumg%C3%BCter/Ladeger%C3%A4te/b6-cms.zip/Hauptplatine.wmf?as=SVG)

As for flashing it, no idea, maybe someone with a JLink can get it to work since SWD seems to get disabled after boot
My stlink does not want to work with it, even with hw reset it gets chipid 0
Pinout for the connector on top is `5v gnd clk dio gnd p51 p50`

diggit, 2023-12-20:
Sorry for lack of updates, but I am giving preference to other projects at this moment. Once I get back to cheali, I'll rather put my effort into #338, as this HW does not even have differential op amps for cell measurement, to it kinda sucks.

DSchndr, 2023-12-21:
@RadioControl there is no calibration menu
But there would be enough space to patch one in
These are pretty much all menus
[image]

DSchndr, 2023-12-23:
@diggit how much accuracy do you want? For me at least the 6mv/step are fine considering the price...
Analog mux is in the chip... so why an external one? 

What is missing would be an adc linearity graph, to know how the adc measures across its voltage range.
Someone with a programmable psu, some time and coding skills might do that.

As for writing code, an arm keil package exists from which the header files and HAL can be extracted.
Flashing it works too over openocd, i've made some scripts for doing that:
https://github.com/DSchndr/charge-me-up/

Btw: check the charge / discharge current of the device, on mine the current is halved which means that either someone forgot a *2 in the code or between developing and manufacturing it someone changed some components...

diggit, 2023-12-23:
@DSchndr You have to subtract 2 ADC channels to get cell voltage so the worst case deviation is 12 mV. Probbly not that bad.
external mux has no benefit, but diff amps do.

I am aware of HAL availability and everything.
Feel free to work on porting cheali to this MCU.

Regarding wrong currents, did you calibrate your charger?

Barrel0Memes97, 2024-02-22:
Any update for this clone? (CMS32L051) Is it flashable with the current release?

agaletski, 2024-03-18:
Buying china clones like this we never know what surprises will arrive this time. I haven't checked currently possible variants, just thought -who will spent time to design clone on different CPU while Atmega32 is so inexpensive. And now I'm one more disappointed  owner of "Imax B6  80W" charger (B6-CMS V12 board, based on CMS32L051 cpu)
Frankly, hardware is not bad and it looks like working but current software makes this device use inconvenient or even useless. 
The main function CHARGING works correctly (current and voltage are  close to selected) but mAh counter become zeroed just after charge finished. In DISCHARGE mode there is surprise. Real discharge current is half of selected. When you set 0.1A current device will not discharge battery at all. Capacity counter in discharge mode counts correct values, it will show 200mAh after hour discharge on 0.4A discharge set on device (the real disch. current is about 0.2A). But the capacity counter become zeroed when discharge process ended.

Is any further progress making this device usable ?
If no luck with porting software to CMS32L051, why not replacing this CPU with  Atmega32 ? 
I just though if CPU replacing is possible solution?

Barrel0Memes97, 2024-03-18:
> Is any further progress making this device usable ? If no luck with porting software to CMS32L051, why not replacing this CPU with Atmega32 ? I just though if CPU replacing is possible solution?

Replacing the CPU would be amazing if the shamanic supported it but I have no idea. Additionally this CPU seems to rarely have opamps integrated into the shamanic so its still less useful even with a CPU seap. Long-term I'll probably just buy a better charger instead of this but it would still be nice if cheap Chinese products like this were all identical.

greekuser, 2024-04-24:
Hi there, i bought a b6 v3 charger from AliExpress without knowing anything about different versions of MCUs. So the title said Imax b6 but i received a Sevenchip b6 one with  a CMS32L051. After searching i was disappointed to find out that i couldn't flash the cheali firmware. Anyway, i could return it but  the charger is working, charging well, with a little deviation of 0.05-0.08mV lower. I can leave with that by adjusting the termination voltage just lower. But the main reason i bought the charger is for the storage function which i cant figure out how its working or if its working at all in this firmware. I know its all matter of code here. So when i choose storage mode(lipo,liion,lihv doesnt matter) with 0.3A, if the battery voltage is more than 3.65v, it starts to discharge with 100ma until 3.65v and then starts charging with 300ma untill full charge, not 3.85v. i have to say that i have only 1s batteries connected to main discharge port and not using any balance port. So either firmware is crappy or every function is supposed to work with the balance port connected and 1s charging/discharging/storage without balance lead is not implemented.
Any hint on that? I will be perfectly happy with the charger if i solve the storage mode problem.

Omino870, 2024-04-24:
I have a trouble with charge/discharge current (uncorrect capacity counter too). For me discharge current meter resistor must to be 0,24Ohm, charge current meter resistor 0,047Ohm. Anybody have a simile problem ?

agaletski, 2024-04-24:
I don't think the device has discharge|charge resistor issue but software is not tested and wrongly set discharge current. As I described in my earlier message, discharge current is only half in fact of value which you specify in setting. Capacity is measured according to real current (half of initialy set value) The worst thing is measurment become zeroed after process finished. You will not know needed values if  not observe the end of process. Hardware is not bad in fact but software is completely shity. 
I hope some one find the way to compile alternative firmware for it.

Omino870, 2024-04-24:
In my device this two resistors was crosschanged. So you have 0,5R resistor in your device. If replace with 0,24R current will up to setted. I just ordered the new resistors, it need some days for delivery. Then i check real charge/discharge current to correspond with settings.

diggit, 2024-04-24:
@agaletski @Omino870 @greekuser please stay on topic. This issue is here to track the progress on CMS32L052 support in cheali.

greekuser, 2024-04-25:
@diggit 
Yes you are correct that it is off topic, however i would prefer that if you indeed know the answer to my question that you would first write it, and then of course write about not going off topic.
Anyway, i wont make another post about it and  in case anyone else seeking about the same as me, i decided to make a diy balance cable for my batteries. Well the voltage readings from the balance cable are EXACTLY as my multimeter and every function now works as it should. Storage mode finishes at 3.85v. So my Sevenchip CMS32L051 is spot on calibrated.

matiaschamu, 2024-12-30:
I recently acquired one of these chargers with a CMS32L051 and the truth is that the hardware doesn't seem that bad to me and the 12mv error is not very serious either.
I think we should try to integrate this CPU into this project because it is incredible how many chargers are on sale and people buy them for their good price vs their features. If you help me, I can spend some time trying to integrate this CPU into the project, I think it would be a fantastic idea. I received an STLink a week ago, I'm going to try connecting and trying to make a backup of the existing firmware to save it and start testing. I have a lot of experience in C++ with ESP32, Atmel and Microchip, but I never did anything with a CMS chip. That's why I tell you that if you help me get started I can help you a lot with the progress of the project.
Sorry my English is not very good :)

cifti, 2025-02-19:
Is there any progress on this issue. I'm very interested if someone can move forward to solve it.

Nadarino, 2025-10-17:
I received one of these units unknowing the rabbit hole these clones entails.  I'm not satisfied with this board/CPU for my use case with NiMH.  I may have melted the discharge nmos U9 and RD resister in my testing/playing.  What I can say is the current measurement and actual current discharge are not the same.  It varies, but what is displayed on screen is ~double what the device is actually sinking.  The current counter on the device seems to be time based at a rate the screen displays.  So you'd be getting half as much charge as you think you are when discharging.  

Example for NiMH set to 1.0A discharge rate, it was measuring 0.4-0.5A's actually being discharged.  About 0.2A was close to the same and 0.1A was 0.3A being discharged.   I susspect a better tuned RD resistor may help calibrate the current discharge

Charging doen't seem to suffer the same issue. its pretty decent at 1-10% was what was observed.   Also when charging it will ramp up to the set value drop to zero and ramp again 2 times before holding stead at the set rate.  So charge counting appears to be more accurate when charging.  

The voltage seemed close it was with in 100mV if not closer but again it wasn't something I had dug into quite yet before blowing it up a little bit.  

Edit: I just realized there was a bunch of hidden content...I don't git much please forgive