stawel/cheali-charger #341: Feature Request: Is it possible to add Iggnus features to this firmware?

bladyle, 2023-05-31:
Hi I'm following that there is new development happening here after few year of pause witch is very good because this charger is still very popular and good enough for the most of the hobbyist. 
Few years back I have flashed mine with[ Iggnus fork  (https://github.com/Iggnus)](https://github.com/Iggnus/cheali-charger-i1) and I'm very happy because of some of his features he implement - mainly on-the-fly adjustments and changing parameters on already stored profile without making new one . 
Can you check it out and maybe add some of them to this firmware? I think this features are very good and make this charger more comparable with current more expensive chargers with color screens and etc. especially now when this firmware is still in updated and maintained.

RadioControl, 2023-06-01:
Yes, that would be very nice! I'm interested too...

rn777, 2023-08-25:
I support. I'm also interested in extending the functionality of the device.

synthkeywizard, 2023-09-05:
I'm very interested too. It is very good to have on-the-fly adjustments.