stawel/cheali-charger #342: Bantam bc-6 AC/DC is it possible?

fu8765, 2023-07-28:
I believe the bantam brand is the original of these chargers. Is this programmable using their USB interface?

diggit, 2023-07-28:
Hi,
there are many "clones" of this charger with various internals. It's not possible to generalize them by model.
I've never seen one with USB/DFU/VCOM bootloader.

fu8765, 2023-08-02:
My understanding is that Bantam is the original target than the many clones

diggit, 2023-08-02:
Target of who/what? Target of cheali is to provide advanced, open source FW for hobby chargers. No matter who manufatured it or what is printed on it. Anyway, have not seen any USB flashable charger in range of chargers, that are cheali compatible. That is *my* experience.

diggit, 2024-02-28:
Hi, sorry it my response sounded rude. I am not sure what was the first target. At this moment, target is almost any clone (as in general concept and schematic) with supported MCU. Cheali is not affiliated with any brand.

fu8765, 2024-07-07:
No worries, and sorry for the extremely late response. 
I finally opened up the charger and have a row of 6 terminals that could be where to hook up the programmer, but there no mention of what's what.