stawel/cheali-charger #343: DC power supply mode

synthkeywizard, 2023-09-05:
Is it possible to add a power supply mode in addition to led mode? Sometimes it is very useful to have a regulated DC source in the same box, when you can set a necessary voltage and current limit. Currently in led mode it works only as constant current and disables output when set voltage reached.

diggit, 2023-09-05:
Hi, did you try to search for this topic here?
Look into: #163 #148

synthkeywizard, 2023-09-05:
Thank you, I didn't seen these topics before. 
Currently, the LiPo charging mode works the way we need, with only one difference - it disables output when current is lower than 10%. If it is possible to deactivate auto-disabling output and make possible to set voltage and current manually - it would be mode we are looking for.

Jio88, 2024-12-04:
https://www.youtube.com/watch?v=RPLZgNXz5b4

focus time  1:50    

Power mode who know about this firmware?