stawel/cheali-charger #345: Any currently available chargers (December 2023) that support this firmware?

victorhooi, 2023-12-05:
I had a look through the hardware, and it seems like many of the models mentioned are discontinued:

[image]

[image]

Does anybody know of any currently available hardware, that supports this firmware?

Thanks!

sustmi, 2024-02-13:
I got this one https://www.aliexpress.com/item/32800729065.html (Color: "Blue B6 80W No AC" and Plug Type: "EU plug").

[image]

The description says it has "ATMEGA32L" and I really got one with ATMEGA32 and successfully use it with [cheali-charger-Dual-Power-B6AC-80W-RC_2.02-e10.3.12-20230523_atmega32](https://github.com/stawel/cheali-charger/blob/master/hex/unstable/cheali-charger-Dual-Power-B6AC-80W-RC_2.02-e10.3.12-20230523_atmega32.hex) firmware.

sustmi, 2024-02-13:
I also found this offer https://www.amazon.de/-/en/RUIZHI-Charger-Professional-Balance-Li-Ion/dp/B08NJXYKGL 

[image]

They mention "ATMEGA32L" in profile picture but I did not try to order this one.