stawel/cheali-charger #350: Imax b6 new 1.8 version board

Vasud23, 2024-02-26:
Hi Pawel,

Can I get correct hex for the below board.

Notice that PCB on the upper right corner has an additional chip, 
1. Is an LM2904.
2. Pin 13 from 74HC4051 it doesn't connect to anything.
3. Pin 14 and 15 from 74HC4051 connected as the schema.

The charger is probably an improved version.

[image]
[image]
[image]

Thanks,
Vasu

diggit, 2024-02-26:
Is your charger capable of 80W? If so, use cheali-charger-Dual-Power-B6AC-80W

Vasud23, 2024-02-27:
I’m not sure, how we can check that?

diggit, 2024-02-27:
It is usually written on the casing of the charger.

Vasud23, 2024-02-28:
Yeah 80w, so I flash the charger with cheali-charger-Dual-Power-B6AC-80W. 

In expert Danger, there is no options for (Vb0pin, Vb1pin, Vb2pin). How I can calibrate my charger with this version?

diggit, 2024-02-28:
There is no such option for 80 W variant, because there is no need for it. Only simplified circuitry in most 50 W chargers needs some additional calibration steps. 80 W chargers usually have differential amplifers on *all* balancer channels.

Vasud23, 2024-02-28:
Ok, but I’m unable to set 2A for discharge, why?

Vasud23, 2024-02-29:
Yeah ok got it, thanks man

closed 2024-02-29 by Vasud23