stawel/cheali-charger #351: Temp issue LM35 - Dual-Power-B6AC-80W

Vasud23, 2024-03-16:
Hi Pawel,

When I connect temp sensor LM35, my charger showing incorrect values like it’s going up and down fast, how to fix this?

I’m using charger-Dual-Power-B6AC-80W firmware - 

80w charger.

Thanks,
Vasu

diggit, 2024-03-17:
Hi,
check your UART settings https://github.com/stawel/cheali-charger/blob/master/docs/settings/settings.md#uart-settings
Set it to "disabled" and there is also option whether temperature and uart pin are separate, pick that they are **not** separate pin.

> it’s going up and down fast
Please be more specific. How significant and how rapid are these changes? eg. "10˚C/s jumping about 30˚C peak/peak"
Is there some kind of pattern? e.g. "same issue repeating every 5 seconds"

Vasud23, 2024-03-17:
It’s "disabled" and there is no option for “temperature and uart pin are not separate pin.”

It’s going up 0C to 225C in milliseconds and then again down to somewhere in 100C its dynamically changing.