stawel/cheali-charger #357: Feature request: battery capacity check function

bladyle, 2024-10-17:
Last couple of days I'm searching for tester to check some old 18650 batteries and cant find ready to go solution for fair price.
What I want is :
- charge the battery to 100% 
- discharge the battery to cut off voltage
- charge the battery to storage voltage
- final screen that display total capacity measured on discharge cycle. 
Is this possible with current firmware and this charger at all?

diggit, 2024-10-19:
Hi, there is a capacity check feature, but I think it finishes on the full charge voltage rather than storage voltage.
Also I think it does multiple cycles. You can see charge/discharge counter of each cyvle during and also in the end.
I you are interested in detals, I can try it out or look into the code.

bladyle, 2024-10-21:
Yes I somehow missed that we have it because it was not listed as feature in main page. But as you said it finishes the cycle with full charge and not in storage voltage that should be better feature I think.