stawel/cheali-charger #360: IMAX B6 06 AC fail - I need help!

DGNLTD, 2025-05-21:
IMAX B6 06 AC charger on the box, IMAX RC on the board. Processor atmega32. The board is similar to the IMAX B6 - original, but there are squares on the screen. I reflashed the backup - the squares remained. What could be the problem? I need help!

[image]
[image]
[image]
[image]

diggit, 2025-05-21:
Hi,
which file you flashed? People often download the hex file incorrectly. Can you show several lines of the file you flashed?

MCU flash readout was probably locked, thus your backup in invalid. Look inside the backup hex, if you see a repeating pattern in the whole file, it is useless.

DGNLTD, 2025-05-21:
I was the last to flash this file - cheali-charger-Turnigy-MAX80W-7A_2.00-e10.3.12-20161202_atmega32.hex

```
:100000000C940F060C9437060C9437060C943706A4
:100010000C946C090C9437060C9437060C94370634
:100020000C9437060C94D0090C949C090C94A709E5
:100030000C9437060C94580A0C948D0A0C943706CD
```

 I still have a bakup like this.

````
:2000000000000101020203030404050506060707080809090A0A0B0B0C0C0D0D0E0E0F0FF0
:2000200010101111121213131414151516161717181819191A1A1B1B1C1C1D1D1E1E1F1FD0
:2000400020202121222223232424252526262727282829292A2A2B2B2C2C2D2D2E2E2F2FB0
```

I even checked the backup, and when I didn't see the FF there, I decided that it was read correctly.

Upd. Turning on the device again now, I saw that the firmware works and offers to reset the eeprom. The buttons also work correctly, I will try calibration.

diggit, 2025-05-22:
Glad to hear that cheali is running. BTW you flashed the incorrect FW, so physical limits are nore not properly set and you damage your charger. In your case, I go with `unstable/cheali-charger-Dual-Power-B6AC-80W-RC_2.02-e10.3.12-20230523_atmega32.hex`